xblax/flashforge_ad5m_klipper_mod #277: MCU 'mcu' shutdown: Move queue overflow

CoreyOLi, 2024-12-30:
got about 95% through a 2.5 hour print job, snippet from log:

Dumping receive queue 100 messages
Receive: 0 18663.493543 18663.489493 16: seq: 1f, thermocouple_result oid=6 next_clock=3932480000 value=16842 fault=0
Receive: 1 18663.600533 18663.580645 15: seq: 16, analog_in_state oid=16 next_clock=3946880000 value=26860
Receive: 2 18663.773607 18663.655794 16: seq: 1a, thermocouple_result oid=4 next_clock=3972800000 value=1245184 fault=0
Receive: 3 18663.793585 18663.655794 16: seq: 1a, thermocouple_result oid=6 next_clock=3975680000 value=16842 fault=0
Receive: 4 18663.900625 18663.655794 15: seq: 1a, analog_in_state oid=16 next_clock=3990080000 value=26863
Receive: 5 18663.981315 18663.949655 11: seq: 1b, clock clock=3959518825
Receive: 6 18663.995593 18663.976626 18: seq: 10, tmcuart_response oid=2 read=b'\n\xfa\xef-\x80"\x02\x08\xb8\xca'
Receive: 7 18664.004749 18663.998558 18: seq: 13, tmcuart_response oid=0 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 8 18664.014182 18664.006673 18: seq: 14, tmcuart_response oid=1 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 9 18664.023233 18664.016372 18: seq: 15, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 10 18664.031679 18664.025470 18: seq: 16, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 11 18664.041260 18664.034711 18: seq: 17, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 12 18664.073563 18664.066138 16: seq: 19, thermocouple_result oid=4 next_clock=4016000000 value=1323008 fault=0
Receive: 13 18664.093582 18664.082464 16: seq: 1b, thermocouple_result oid=6 next_clock=4018880000 value=16842 fault=0
Receive: 14 18664.200598 18664.174914 15: seq: 11, analog_in_state oid=16 next_clock=4033280000 value=26854
Receive: 15 18664.373567 18664.367659 16: seq: 1e, thermocouple_result oid=4 next_clock=4059200000 value=1167360 fault=0
Receive: 16 18664.393576 18664.367659 16: seq: 1e, thermocouple_result oid=6 next_clock=4062080000 value=16842 fault=0
Receive: 17 18664.500498 18664.495594 15: seq: 16, analog_in_state oid=16 next_clock=4076480000 value=26852
Receive: 18 18664.673546 18664.667668 16: seq: 12, thermocouple_result oid=4 next_clock=4102400000 value=1245184 fault=0
Receive: 19 18664.693579 18664.667668 16: seq: 12, thermocouple_result oid=6 next_clock=4105280000 value=16842 fault=0
Receive: 20 18664.800564 18664.667668 15: seq: 12, analog_in_state oid=16 next_clock=4119680000 value=26852
Receive: 21 18664.973688 18664.942539 16: seq: 1d, thermocouple_result oid=4 next_clock=4145600000 value=1245184 fault=0
Receive: 22 18664.987703 18664.956315 11: seq: 13, clock clock=4104446152
Receive: 23 18664.993529 18664.962103 16: seq: 15, thermocouple_result oid=6 next_clock=4148480000 value=16842 fault=0
Receive: 24 18665.001917 18664.992537 18: seq: 18, tmcuart_response oid=2 read=b'\n\xfa\xef-\x80"\x02\x08\xb8\xca'
Receive: 25 18665.010721 18665.003896 18: seq: 1a, tmcuart_response oid=0 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 26 18665.020441 18665.012862 18: seq: 1b, tmcuart_response oid=1 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 27 18665.029906 18665.022492 18: seq: 1c, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 28 18665.038739 18665.032511 18: seq: 1d, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 29 18665.046556 18665.040689 18: seq: 1e, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 30 18665.100514 18665.090639 15: seq: 12, analog_in_state oid=16 next_clock=4162880000 value=26862
Receive: 31 18665.273630 18665.263294 16: seq: 1e, thermocouple_result oid=4 next_clock=4188800000 value=1245184 fault=0
Receive: 32 18665.293558 18665.276885 16: seq: 1f, thermocouple_result oid=6 next_clock=4191680000 value=16842 fault=0
Receive: 33 18665.400525 18665.387339 15: seq: 15, analog_in_state oid=16 next_clock=4206080000 value=26854
Receive: 34 18665.573519 18665.553369 15: seq: 10, thermocouple_result oid=4 next_clock=4232000000 value=1245184 fault=0
Receive: 35 18665.593490 18665.586393 15: seq: 12, thermocouple_result oid=6 next_clock=4234880000 value=16842 fault=0
Receive: 36 18665.700524 18665.677800 14: seq: 18, analog_in_state oid=16 next_clock=4249280000 value=26855
Receive: 37 18665.873779 18665.701267 15: seq: 19, thermocouple_result oid=4 next_clock=4275200000 value=1323008 fault=0
Receive: 38 18665.893585 18665.701267 15: seq: 19, thermocouple_result oid=6 next_clock=4278080000 value=16842 fault=0
Receive: 39 18665.990010 18665.958657 10: seq: 19, clock clock=4248779877
Receive: 40 18666.000545 18665.973161 14: seq: 1d, analog_in_state oid=16 next_clock=4292480000 value=26857
Receive: 41 18666.004496 18665.973161 18: seq: 1d, tmcuart_response oid=2 read=b'\n\xfa\xef-\x80"\x02\x08\xb8\xca'
Receive: 42 18666.013969 18666.006368 18: seq: 1f, tmcuart_response oid=0 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 43 18666.023588 18666.016054 18: seq: 10, tmcuart_response oid=1 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 44 18666.034079 18666.026705 18: seq: 12, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 45 18666.043354 18666.036081 18: seq: 13, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 46 18666.051450 18666.045162 18: seq: 14, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 47 18666.173515 18666.152546 15: seq: 1b, thermocouple_result oid=4 next_clock=23432704 value=1323008 fault=0
Receive: 48 18666.193497 18666.188412 15: seq: 1e, thermocouple_result oid=6 next_clock=26312704 value=16842 fault=0
Receive: 49 18666.300557 18666.277945 14: seq: 15, analog_in_state oid=16 next_clock=40712704 value=26853
Receive: 50 18666.473526 18666.456714 15: seq: 11, thermocouple_result oid=4 next_clock=66632704 value=1245184 fault=0
Receive: 51 18666.493557 18666.482754 15: seq: 12, thermocouple_result oid=6 next_clock=69512704 value=16842 fault=0
Receive: 52 18666.600451 18666.594721 14: seq: 19, analog_in_state oid=16 next_clock=83912704 value=26854
Receive: 53 18666.733601 18666.659473 16: seq: 1e, stats count=591 sum=2979396 sumsq=73285628
Receive: 54 18666.773549 18666.659473 15: seq: 1e, thermocouple_result oid=4 next_clock=109832704 value=1245184 fault=0
Receive: 55 18666.793534 18666.659473 15: seq: 1e, thermocouple_result oid=6 next_clock=112712704 value=16842 fault=0
Receive: 56 18666.900669 18666.659473 14: seq: 1e, analog_in_state oid=16 next_clock=127112704 value=26848
Receive: 57 18667.011079 18666.979042 10: seq: 10, clock clock=100850644
Receive: 58 18667.025507 18666.992647 18: seq: 15, tmcuart_response oid=2 read=b'\n\xfa\xef-\x80"\x02\x08\xb0\xcd'
Receive: 59 18667.035155 18667.027923 18: seq: 18, tmcuart_response oid=0 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 60 18667.044929 18667.037417 18: seq: 19, tmcuart_response oid=1 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 61 18667.055354 18667.047253 18: seq: 1b, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 62 18667.064316 18667.057872 18: seq: 1c, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 63 18667.072564 18667.066266 18: seq: 1d, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 64 18667.073544 18667.066266 15: seq: 1d, thermocouple_result oid=4 next_clock=153032704 value=1323008 fault=0
Receive: 65 18667.093509 18667.066266 15: seq: 1d, thermocouple_result oid=6 next_clock=155912704 value=16842 fault=0
Receive: 66 18667.200490 18667.193958 14: seq: 15, analog_in_state oid=16 next_clock=170312704 value=26838
[... 11 lines cut ...]
Receive: 78 18668.048479 18668.040999 18: seq: 1f, tmcuart_response oid=1 read=b'\n\xfa\xef-\x804\x02\x08\xa0\xd9'
Receive: 79 18668.058242 18668.050672 18: seq: 10, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 80 18668.067736 18668.060330 18: seq: 11, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 81 18668.077597 18668.070550 18: seq: 12, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 82 18668.100528 18668.070550 15: seq: 12, analog_in_state oid=16 next_clock=299912704 value=26835
Receive: 83 18668.273569 18668.254100 16: seq: 1e, thermocouple_result oid=4 next_clock=325832704 value=1245184 fault=0
Receive: 84 18668.293533 18668.286372 16: seq: 10, thermocouple_result oid=6 next_clock=328712704 value=16842 fault=0
[... cut ...]
Receive: 96 18669.062863 18669.055766 18: seq: 16, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 97 18669.072753 18669.064991 18: seq: 18, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 98 18669.081744 18669.074975 18: seq: 19, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 99 18669.143675 18669.140069 12: seq: 1e, shutdown clock=407941729 static_string_id=Move queue overflow
Transition to shutdown state: MCU 'mcu' shutdown: Move queue overflow
Dumping gcode input 0 blocks
Dumping 20 requests for client 3041908496

KaruroChori, 2024-12-30:
This is the case when the sliced model has a complexity, a number of moves and a timeframe they are scheduled exceeding the capabilities of the underlying hardware.  
You have to either slow down your slicing profiles, or reduce the resolution of arc segmentation for example.

closed 2024-12-30 by KaruroChori

CoreyOLi, 2024-12-30:
That's good news, because I am really enjoying this firmware and would hate to have to revert...  I'm fairly new to 3d printing and I'm using Orca Slicer, could you give me a quick heads up on where I make that adjustment?

KaruroChori, 2024-12-30:
It depends on whether you are generating G2/G3 arc moves or not in your slicing profile.
If so, the only option would be to make them more coarse or decrease the global speed of your printer. The first one is done by changing the klipper config file (https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/printer_configs/printer.base.cfg), but it is already set to 1mm which is quite coarse, so I would not touch that.
You can globally decrease speed either by changing the global value via fluidd/mainsail as a quick fix, or re-slice your model.

If you are not generating G2/G3 moves, then it is up to the slicer to reduce the resolution of arc segmentation. These are the settings you might want to tweak https://wiki.bambulab.com/en/software/bambu-studio/acr-move

CoreyOLi, 2024-12-30:
so when I slice my model I just need to print at a slower speed, is that it?   seems easy enough

CoreyOLi, 2024-12-30:
I'd like to make this a "permanent" change within the firmware, to slow it down...  I'm not seeing anything in the printer config, I see a toolhead speed in settings it's set at 130mm/s for x/y, but I feel that isn't what I'm looking for...

KaruroChori, 2024-12-30:
Every MCU has a cap on how many moves it is able to push in a second. So you either relax time, or you relax the number of moves. Lowering the target speed relaxes time, lowering the slicing resolution of arcs decreases the number of moves.  

Of the two, if you can live with a reduced resolution, the latter is better. Otherwise you will be impacting speed for ALL actions, even simple straight lines.  
Sadly, there is no way to configure a cap in the slicer to directly limit the moves/second and match your hardware's capabilities. That would be massively helpful to avoid this sort of problems.

KaruroChori, 2024-12-30:
https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/dc421d0fdb79d93fe5e96bec652556903dfd808a/printer_configs/printer.base.cfg#L168

You have several global settings in here.

CoreyOLi, 2024-12-30:
okay, I'll dig in and look at the settings, thank you!!!

CoreyOLi, 2024-12-30:
> Every MCU has a cap on how many moves it is able to push in a second. So you either relax time, or you relax the number of moves. Lowering the target speed relaxes time, lowering the slicing [...]

I think I understand,  gcode_arcs resolution is set to 1.0...  What would be a reasonable setting to avoid a potential issue?