xblax/flashforge_ad5m_klipper_mod #281: Hyperlapse not working

rcompton, 2025-01-03:
I can't take timelapses and see this in my moonracker log. 

2025-01-03 05:59:19,673 [shell_command.py:_check_proc_success()] - Command (wget http://xxx:8080?action=snapshot -O /root/printer_data/timelapse/images/frame000001.jpg) exited with return code 1
2025-01-03 05:59:19,674 [timelapse.py:newframe()] - getting newframe failed: wget http://xxx:8080?action=snapshot -O /root/printer_data/timelapse/images/frame000001.jpg

I think wget needs quotes

kylerjensen, 2025-02-07:
Offending line of code is here: https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/dc421d0fdb79d93fe5e96bec652556903dfd808a/prebuilt/moonraker-plugins/timelapse/timelapse.py#L473

kylerjensen, 2025-02-07:
I had this same problem. I was able to fix it by adding an extra forward slash (/) between 8080 and the question mark character in my webcam URLs. I also learned that I had to use the printer's IP address for the stream URL, but localhost worked fine for the snapshot URL as long as I used the UV4L-MJPEG service. Here is a screenshot of my working configuration:

[image]

closed 2025-09-16 by xblax