xblax/flashforge_ad5m_klipper_mod #287: Filament runout guppy screen

DocJ-GA, 2025-01-18:
I had a filament run out detection happen today.  It went great, job was paused and i was notified.  I used the mainsail screen to resume.  The issue was on the guppy screen.  It had a dialog on it that would not go away.  Did not affect the print but essential had a message that it ran out with a button that said xyz.  Clicking the button did nothing and the only way to clear the dialog was to turn the printer off and on.  A firmware restart did nothing, as expected and I could not find a way to restart the guppy screen.

konradzawadka, 2025-01-22:
Retested today, and it looks that everything is working as expected.

luckyPipewrench, 2025-01-23:
This happened to me a several times. Restarting the machine from the switch has been the only way to resolve once the Guppy Gui freezes. A way to reboot Guppy from the web interface would be a nice feature.

rudik99, 2025-04-05:
I experience the same thing and reboot is the only thing that fixes it
After pressing resume the word resume starts changing to random letters and characters

jestepp, 2025-05-02:
Still happens on the latest release, looking into this myself. Looks dead here? i am a noob so i wont get far!

xblax, 2025-09-16:
Guppy Screen variant is labeled as "alpha", not in a fully functional state unfortunately. You can retry with the latest release, but not guarantee that it works properly.

I'm closing the GuppyScreen issues, since it's unlikely they will be worked at by someone and upstream development has also stalled. I added some information about Guppy Screen "broken" state to [Known Issues](https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/LINUX.md), so that it's clear it can be used as-is but will not be worked on at the moment.

closed 2025-09-16 by xblax