xblax/flashforge_ad5m_klipper_mod #296: will this work with new ADX color?

AbitNutzT00, 2025-02-07:
I'm all about putting Klipper on all my machines. I've done my Sovol SV08 and  Flsun V400. Will this mod work for the new AD5X multi color machine?

spilledfries, 2025-02-15:
if the ad5x reuses the same mainboard as the ad5m (which it probably won't), then it could possibly work, but trying to reverse engineer the multi filament system would take some work, and considering this repo hasn't been updated in almost a year, you might as well look elsewhere, or try to reverse engineer the ad5x's firmware and such.

max-hasky, 2025-03-29:
AD5X support is in [zmod](https://github.com/ghzserg/zmod).
KlipperMod can also be ported to AD5X, but you need to rebuild BuildRoot for mips and change the paths.

xblax, 2025-09-16:
It could be ported for sure, but it's not planned. None of the persons that develop this mod owns AD5X.