xblax/flashforge_ad5m_klipper_mod #305: New Flashforge firmware not allowing klipper install

Simmondsp, 2025-03-07:
Hi all

has anyone tried installing klipper since the new flashforge update around 3 weeks ago? 

I am no longer able to install klipper on the machine now. 

Thanks for help in advanced.

BardinGuitars, 2025-03-07:
Yes. Works well.

grahamwilliams, 2025-03-23:
I also can't install. I've got FAT32 formatted USB with the zip file on the drive as well as a FAT32 formatted USB with the files unzipped. On 3.1.5 it does not begin the upgrade with either USB drive

SeanSeenBean, 2025-03-24:
> I also can't install. I've got FAT32 formatted USB with the zip file on the drive as well as a FAT32 formatted USB with the files unzipped. On 3.1.5 it does not begin the upgrade with either USB drive

I am in the same boat, was using an ext4 drive at first but even with FAT32 zipped or unzipped it doesn't work. I can't get it to install even an older official firmware. I'm not sure if it's a bug with 3.1.5 or if it is intentional. I do regret updating the printer immediately after buying it yesterday though. If I find a solution to the issue I'll comment it here.

dmi37in, 2025-03-28:
I just installed the mod on the latest firmware 3.1.5. I just put the firmware file in .tgz format and everything worked. My flash card has a FAT32 system.

SeanSeenBean, 2025-03-28:
> I just installed the mod on the latest firmware 3.1.5. I just put the firmware file in .tgz format and everything worked. My flash card has a FAT32 system.

Huh, tried a different flash drive today, freshly formatted to FAT32 and now it is installing. Thanks. I guess if anyone has issues installing just keep trying other drives.

SeanSeenBean, 2025-03-28:
> I just installed the mod on the latest firmware 3.1.5. I just put the firmware file in .tgz format and everything worked. My flash card has a FAT32 system.

Huh, tried a different flash drive today, freshly formatted to FAT32 and now it is installing. Thanks. I guess if anyone has issues installing, just keep trying other drives.

grahamwilliams, 2025-04-01:
Okay there has to be something I'm doing wrong. I've been through a dozen USB drives, formatting them both through Finder right click and disk utility as MS-DOS FAT (FAT32) on MacOS (and once on windows!)

Does anyone have any ideas wtf is going on??

MatthewPricePhd, 2025-04-01:
> Okay there has to be something I'm doing wrong. I've been through a dozen USB drives, formatting them both through Finder right click and disk utility as MS-DOS FAT (FAT32) on MacOS (and once on windows!)
> 
> Does anyone have any ideas wtf is going on??

Find the smallest drive you can… I had to use a 512mb novelty drive I found.  Even if you format FAT32, on windows 11 at least, when I tried 1gb or larger drives it didn’t work for me.  I don’t know if windows sees the larger size and does something with xFAT or what. Might just be my work computer but the smaller/older driver worked first try.

grahamwilliams, 2025-04-01:
Figured it out - it wasn't drive size, though I suspected that. 

It was the f*%&ing Flaccid Flexible c*%&ing Cable again - I was wondering why my camera was dropping in and out. I trimmed a bit off the end of the cable, re-inserted it, boom install. 

I love this little printer but the son of a bleach at FF who decided to have a consumer facing part this fragile AND had their own mod kit be a torture test for said cable.... well, hell has a luxury resort with their name on it. Nameless engineer: I hate you so very much. Not really. But seriously, my dude.

closed 2025-09-16 by KaruroChori