xblax/flashforge_ad5m_klipper_mod #308: Circulation servo position incorrect

AR1972, 2025-03-15:
the door on my AD5M Pro was timed to be fully closed at 115 degrees, while clipper mod has the door timed to be fully closed at 90 degrees
this causes a grinding noise when Klipper mod starts, if you encounter the grinding noise, hit the AIR_CIRCULATION_EXTERNAL macro and edit printer.base.cfg find inital_angle:90 and change it to initial_angle:115 then open macros-pro.cfg and change all instances of ANGLE=90 to ANGLE=115 then save and restart.

Because of the way servos work the servo will be damaged/destroyed fighting to close the closed door even more. if your servo does melt down, then the SG90 servos can be made to work by swapping the red and yellow wire or carefully opening up the servo being careful to not open the gearbox and swapping the pig tail. 

[servo my_servo]
pin:PC9
maximum_servo_angle: 180
minimum_pulse_width: 0.0005
maximum_pulse_width: 0.0025
initial_angle:115
#initial_pulse_width:

[image]

fernplant, 2025-09-26:
I have Pro version and have not encountered this issue

AR1972, 2025-09-28:
That's weird, I pulled the stock config off the printer, I wonder if they installed the door wrong and then just tuned it out at the factory?

`[mcu eboard]
serial: /dev/ttyS1
baud: 230400
restart_method: command

[mcu]
serial: /dev/ttyS5
baud: 230400
restart_method: command

[stepper_x]
step_pin: PC1
dir_pin: PC0
enable_pin: !PA1
microsteps: 16
rotation_distance: 40
endstop_pin: !eboard:PA15
position_endstop: 110
position_max: 125
position_min: -125.0
homing_speed: 50
homing_retract_dist: 10
homing_retract_speed: 25
#homing_positive_dir: false

[stepper_y]
step_pin: PA3
dir_pin: PA2
enable_pin: !PA6
microsteps: 16
rotation_distance: 40
endstop_pin: !eboard:PB11
position_endstop: 110
position_max: 125.0
position_min: -125.0
homing_speed: 50
homing_retract_dist: 10
homing_retract_speed: 25
#homing_positive_dir: false

[stepper_z]
step_pin: PC4
dir_pin: PA7
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: PB15
position_endstop: 220
position_max: 230
position_min: -10
homing_speed:20
#homing_positive_dir: false 
homing_retract_dist: 5  
homing_retract_speed: 10

[heater_bed]
heater_pin: PB9
sensor_type: Generic 3950
sensor_pin: PC3
pullup_resistor: 4700
control: pid
pid_Kp: 32.79
pid_Ki: 4.970
pid_Kd: 54.118
#control: watermark
#max_power: 1.0
min_temp: -100
max_temp: 130
[... 229 lines cut ...]
[exclude_object]

#AutoCalibration                   
[bed_mesh]                         
speed: 300                          
horizontal_move_z:5              
move_check_distance: 8             
mesh_min: -105, -105            
mesh_max: 105, 105             
probe_count: 5, 5                  
mesh_pps: 3, 3                     
algorithm: bicubic                 
bicubic_tension: 0.2               
#fade_start: 3
#fade_end: 10
#fade_target: 0
                                  
[probe]                            
pin: !PB4 
z_offset: -0.25       
speed: 5                   
samples: 3
samples_result:average
samples_tolerance:0.100
samples_tolerance_retries:2`

fernplant, 2025-09-28:
Thought that came to mind is maybe yours is a later revision? Not sure