xblax/flashforge_ad5m_klipper_mod #314: Guppy screen Z+ and Z- arrows are mixed up!

BigToyBox, 2025-04-02:
Z+ is UP off the bed. The arrow pointing up should be associated with Z+

JacquesHumule, 2025-04-02:
No it's normal, z- mean hotend needs to go near the bed.
The Z distance is the distance between bed and nozzle

BigToyBox, 2025-04-08:
Wrong! When I push the Z+ the Z moves positive which has ALWAYS meant to raise the nozzle from the bed. The arrow for Z+ should be pointing UP. Up is a positive number and down is a negative number in the real world.

BigToyBox, 2025-04-20:
Update: I switched to the Klipperscreen version. It works SO much better without any lockup/delay issues. 

However there is ONE flaw. At the end of a print I clicked on "save offset" button that appeared at the end of the print. I restarted the same print and what was a (-) Z adjustment before, I had to correct with a (+) adjustment on the fly of the EXACT SAME AMOUNT. AS I said above, the programmers have the z offset "babysteps" graphics and some of the code backwards. Try it yourselves.

xblax, 2025-09-16:
Guppy Screen is labeled as a preview, it has some some issues. I recommend KlipperScreen which also works quite reliably with the latest release.

I'm closing the GuppyScreen issues, since it's unlikely they will be worked at by someone and upstream development has also stalled. I added some information about Guppy Screen "broken" state to [Known Issues](https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/LINUX.md), so that it's clear it can be used as-is but will not be worked on at the moment.