xblax/flashforge_ad5m_klipper_mod #316: camera not showing

css310310, 2025-04-07:
Hi I have been using the guppy screen install for a month now with no issues but yesterday I went to start a print and the camera just shows a blank screen in fluidd and mainsail is there any way to bring the camera back? I have tried to restart the printer and deleted and added the camera again but no luck.I am using the FLASHFORGE Camera and made sure it was plugged in.

[image]

BigToyBox, 2025-04-08:
Yep, mine quit too. This setup worked for a couple of days...

[image]

dmi37in, 2025-04-24:
Hi, simply visit http://YOUR_PRINTER_IP_ADDRESS:8080 for look command help strings

Use **/stream** and **/snapshot** in your case

```
µStreamer v5.51
/state
Get JSON structure with the state of the server.

/snapshot
Get a current actual image from the server.

/stream
Get a live stream. Query params:

key=abc123
The user-defined key, which is part of cookie stream_client, which allows
the stream client to determine its identifier and view statistics using /state.

extra_headers=1
Add X-UStreamer-* headers to the /stream handle (like with the /snapshot).

advance_headers=1
Enable workaround for the Chromium/Blink bug #527446.

dual_final_frames=1
Enable workaround for the Safari/WebKit bug when using option --drop-same-frames.
Without this option, when the frame series is completed, WebKit-based browsers
renders the last frame with a delay.

zero_data=1
Disables the actual sending of JPEG data and leaves only response headers.

The mjpg-streamer compatibility layer:

/?action=snapshot as alias to the /snapshot.

/?action=stream as alias to the /stream.
```

[image]

closed 2025-09-16 by xblax

The-Synthax, 2026-01-17:
The proper solution to this is to use "http://AD5M.local:8080/?action=stream" for the stream URL and the same but =snapshot for the snapshot URL. This will make it IP agnostic and shouldn't require any additional config if the IP of the printer changes or it is connected to a new network.