xblax/flashforge_ad5m_klipper_mod #322: Load cell customization

FaqT0tum, 2025-05-03:
I replaced all the electronics with SKR Pico, RPi, EBB36, hall effect 3144 and FFC breckout. 
I managed to configure everything but I have the problem to do reverse engineering regarding the load cells. 
I managed to trace the pins used on the original mcu (PC12, PD2, PB3, PB4) but they do not match in the printer_base.cfg.

I don't know how to connect the board that manages the load cells to the SKR Pico or rapsberry as I don't know the real use of the pins and communication protocol (spi, uart, i2c, custom, etc). I also discovered that on the load cells board there is an 8pin MCU STC8G sot8, probably processes a part of the values ​​coming from the HX711 and through a protocol such as SPI communicates with the main MCU

FaqT0tum, 2025-05-03:
[image]

FaqT0tum, 2025-05-03:
[image]

FaqT0tum, 2025-05-03:
[image]

FaqT0tum, 2025-05-03:
[image]

this is the board that is giving me a headache

as soon as i solve this, i will post all the useful to replicate it

Mazbury123, 2025-05-07:
Your about there, would love to convert mine, but I cannot work out all of your wiring from the pictures, That's a hx711 loadcell amplifier pcb 4 inputs from the load cells A B C D i think and maybe the input voltage and tx rx output threw that ribbon cable. Hard to tell. where is that loadcell pcb located in the printer, I've only just got my 5m pro. I'm used to building 3d printers from scratch wanted something cheap to print asa and abs in for my printer builds, but hate the lockdown with orca slicer much prefer cura.

FaqT0tum, 2025-05-07:
> Your about there, would love to convert mine, but I cannot work out all of your wiring from the pictures, That's a hx711 loadcell amplifier pcb 4 inputs from the load cells A B C D i think and maybe [...]

I managed to port the load cells, I did some measurements and some tests, the solution was right under my nose. Just use IO-1 as the endstop signal pin. I updated the git with some useful information about it. The load cells board is under the Bed

closed 2025-05-07 by FaqT0tum

Mazbury123, 2025-05-10:
Hi, I'm looking threw your printer.cfg but can not find how you have asigned the correct pin out to the y stop on the pico gpio3. I think you have connected the fm-chk to this which i think is the filament sensor runout on the original pcb. Have you missed this!, or am I missing something or have you not implemented this yet.

Thanks
:-)

FaqT0tum, 2025-05-11:
> Ciao, sto guardando il tuo file printer.cfg ma non riesco a capire come hai assegnato il pinout corretto al finecorsa Y sul pico gpio3. Credo che tu abbia collegato l'FM-CHK a questo, che credo sia il runout del sensore del filamento sul PCB originale. Ti è sfuggito? O mi sfugge qualcosa? O non l'hai ancora implementato?
> 
> Grazie :-)

you are right, the runout implementation is missing. I must have missed it. I'll add it now.
The printer.cfg is what I'm currently using to print and I assure you 100% that it works.

thanks for pointing it out

Mazbury123, 2025-05-14:
Hi, I've been looking at that loadcell pcb, the four holes at the top next to the marked 'ISP' which means in circuit programmer are 'vcc rx tx gnd' this is where that 8 pin microprocessor has been flashed with some code marked as U7, I don't think you will be able to tap into that processor because we don't know how it's been programmed or what protocol they are using, The best bet is to tap into the pins on the HX711 and connect them to the gpio pins on the rasp pi then try and find the correct libs to install and configure, I think they are using some form of a wheatstone bridge in the way those load cells are connected to the HX711. I don't have my printer in bits so I can not trace that printed circut board for you.

FaqT0tum, 2025-05-14:
no need to understand what the mcu does! I managed to make it work as a probe and it is precise. I have been printing with it for 1 week and wrote what is needed here

[](https://github.com/FaqT0tum/FlashForge_A5M_raspberry)

Mazbury123, 2025-05-14:
Will it do a height map in mainsail and level the bed before a print?.

FaqT0tum, 2025-05-14:
yes exactly, just like if it were a BLTouch or pinda. 

in short you do not lose the basic printer function for bed leveling

FaqT0tum, 2025-05-14:
the board on the A5M that manages the load cells (present under the Bed) already manages the H711 by itself via STC8G and uses the pin (IO-1) to output an analog value that goes from 5.0 to 0 volts varying based on the pressure exerted on the Bed. doing so you have 2 possibilities: use the pin as 0/1 by connecting it to an endstop pin using the [probe] function or use an analog pin on the raspberry and manage the analog value with macros so you can do more things with it

Mazbury123, 2025-05-14:
Thanks for the reply, I understand now how you have configured it 0/1 off or on, just took me a while to work out where those 3 wires go that are not connected to anything on your picture, the blue brown and white go to the x stop pins. Got my pico and pi coming tommorrow will upload some pictures as a go. Thanks for all your hard work.
:-)