xblax/flashforge_ad5m_klipper_mod #337: Sporatic thumbnail/metadata loss when transfering gcode files over WiFi on new version

dragos240, 2025-09-26:
I was on the previous version for a while, and while it was buggy, the metadata was almost always transferred correctly. In this new version, the thumbnails and projected print times don't always transfer over when sending over gcode. I don't know what causes this, I just know it usually doesn't transfer that data, but sometimes it does. Anyone else having this issue?

dragos240, 2025-09-27:
It looks like it's related to [this issue in the moonraker repo](https://github.com/Arksine/moonraker/issues/893)

dragos240, 2025-09-27:
The fix [here](https://github.com/Arksine/moonraker/issues/893#issuecomment-2267088889) works, so it's something to do with msgspec or uvloop

cmenard, 2025-09-30:
Thanks! After disabling uvloop, I was able to get all my metadata back.