xblax/flashforge_ad5m_klipper_mod #339: Can't get it to install at all

Lsutton666, 2025-09-26:
I've tried all versions (one at a time) v00.06 beta... Tried renaming... Etc etc    ... So I unzip the file and put it on flash drive? Or do I have to rename and then do that- still nothing... Ive tried zipped, unzipped, renamed etc... after unzipping do I copy all the files to the flash drive or do I have to unzip everything even further please help been trying for awhile now

Lsutton666, 2025-09-27:
> There is not need to rename or unzip anything. Just copy the correct file
> for your printer as you download to a USB flash drive.
> Make sure it's at FAT32 formatted flash drive, not exfat.

Strange.. def reformatted the USB drive to fat32, and tried everything else
.. do I download the source code instead or something?
"Adventurer5M-KlipperMod-v00.06-beta-guppyscreen". I open the drive on my
windows PC and I see chroot, chroot.tar and flashforge_init.sh.   Unless
I'm stupid (which could be the case idk I've tried pretty much everything)

Lsutton666, 2025-09-27:
Strange.. def reformatted the USB drive to fat32, and tried everything else .. do I download the source code instead or something?  "Adventurer5M-KlipperMod-v00.06-beta-guppyscreen". I open the drive on my windows PC and I see chroot, chroot.tar and flashforge_init.sh.   Unless I'm stupid (which could be the case idk I've tried pretty much everything)

xblax, 2025-09-27:
Just copy the "Adventurer5M-KlipperMod-v00.06-beta-guppyscreen.tgz "file to the flash drive, do not unpack it. If you see chroot.tar you have already unpacked it.

But I recommend the klipper screen variant, it's more stable and better tested.